wadpac/oss-dev-webinar-series-pb-field #1: Collection data reading tools: raw data accelerometry

vincentvanhees, 2019-09-04:
Let's try to create an overview of Open Source libraries that aid reading raw accelerometer data. With this overview we may be able to avoid doing the same work twice. The ones I can quickly think of for now are:

- [AGread](https://cran.r-project.org/web/packages/AGread/index.html) for Actigraph data
- [GENEAread](https://cran.r-project.org/web/packages/GENEAread/index.html) for GENEActiv data
- [GGIR](https://cran.r-project.org/web/packages/GGIR/index.html) for various data formats, code written in R/C++
- [biobankAccelerometerAnalysis](https://github.com/activityMonitoring/biobankAccelerometerAnalysis) for Axivity data, code written in Python/Java.

paulhibbing, 2019-09-05:
I'm aware of functions that read activPAL event files in two places:

- [activpalProcessing](https://cran.r-project.org/package=activpalProcessing)
- [Sojourn](https://cran.r-project.org/package=Sojourn)

If I'm not mistaken, the functions are similar, but the function in `Sojourn` may do a better job handling timestamps during daylight saving, with the tradeoff that it requires a `.def` file in the same directory as the `events.csv` file.

TheTS, 2019-09-05:
And another R option for Actigraph binary data (not on CRAN, yet):

- [read.gt3x](https://github.com/THLfi/read.gt3x) for Actigraph gt3x files

and while not raw data, this can do agd files:

- [actigraph.sleepr](https://github.com/dipetkov/actigraph.sleepr) for Actigraph agd files

paulhibbing, 2019-09-18:
@TheTS How familiar are you with `read.gt3x`? I haven't looked closely at it, but I have a few thoughts from my experience and opinion. Of course, these come with the disclaimer that `read.gt3x` may well be better than `AGread`:

* First, my shameless plug: `AGread` also has binary file reading via `AGread::read_gt3x`.
  - I just uploaded a CRAN release (v1.0.0) that uses `Rcpp` and reads ~80% faster than its predecessor (v0.2.2), based on my extremely limited testing.
  - The API is also set up to let you skip parts of the file you're not interested in reading, which can speed things up even more. For example, if you're only interested in primary accelerometer data, you can skip IMU data, battery status updates, capsense data, etc.

* More importantly, I struggled a lot to get the same results from `AGread::read_gt3x` versus the csv output from ActiLife (`*.RAW.csv` and `*-IMU.csv`). After lots of back-and-forth with ActiGraph I've finally got a good grasp on how they handle oddities in the data streams, such as USB connection events and over-/under-sampled IMU packets. Those are potentially things the `read.gt3x` maintainers/users should also be aware of and watch out for.

* `AGread` is also clear on the copyright side with ActiGraph. Their online documentation is open source under MIT, but it was a bit of a headache to figure out how best to comply with CRAN policies.

* Lastly, scope-wise it makes sense to me to have dedicated reading packages for different devices, rather than piecemeal functions scattered across different packages. I can see that `read.gt3x` and `actigraph.sleepr` aren't developed by the same people, but my vote would go for all of those file reading functionalities to live in one package - It wouldn't have to be `AGread`, but you can guess my bias.

**Tl;dr:** I'm curious how `read.gt3x` compares to `AGread`, and I've learned some lessons during development of the latter that I think are worth mentioning for anyone else working with ActiGraph binary data.

martakarass, 2019-09-18:
Terrific to have the AGread::read_gt3x co-author on the thread!

I made attempt to convert GT3X to CSV with the 2 implementations: (1)
read.gt3x::read.gt3x and (2) AGread::read_gt3x. This is ~ 40 minutes of raw
data collected at Hz=100 with idle mode enabled. *I tried to* *emulate the
ActiLife raw data output CSV* I do have for comparison. I failed with both
implementations. Observations:

For read.gt3x::read.gt3x(imputeZeroes = FALSE):

   - Timestamp range I got is reasonable but differs w.r.t actual
   start-download time (whereas start and download times I see in (a) ActiLife
   raw data output metadata, (b) read.gt3x::read.gt3x verbose output - they do
   agree).
   - Sensor idle state values seem not to be handled; that is, I see no
   measurement observations for a few minutes of data in a middle where I
   recall sensor was laying still on a desk.
   - The first minute of raw data, except the 1s shift, do look
   reasonable w.r.t ActiLife raw data output.

For read.gt3x::read.gt3x(imputeZeroes = TRUE):

   -  Timestamp range is off (I see "2020-05-13 06:38:24 GMT" value as
   upper range time value)
   - Still: sensor idle state values seem not to be handled; that is, I see
   no data for a few minutes of data in a middle where I recall sensor was
   laying still on a desk.

For AGread::read_gt3x:

   - Running read_gt3x(verbose = TRUE) failed for me: verbose message
   suggests "Parsing ACTIVITY2 packet" fails (note one would not learn that
   from the error output alone!)
   - Running

read_gt3x(gt3x.destfile, verbose = TRUE,
                  include = c("METADATA", "PARAMETERS", "SENSOR_SCHEMA",
"BATTERY", "EVENT", "TAG", "ACTIVITY", "HEART_RATE_BPM", "HEART_RATE_ANT",
"HEART_RATE_BLE","LUX", "CAPSENSE", "EPOCH", "EPOCH2", "EPOCH3", "EPOCH4",
"SENSOR_DATA"))

(note "ACTIVITY2" is removed)  does complete, but the outcome has the
following attributes only:  "METADATA", "BATTERY", "EVENT", "CAPSENSE",
"PARAMETERS" (no "ACTIVITY" data -> no raw acc data read, right? =C )

Am I mis-using the AGread::read_gt3x? I'd appreciate any hint.

I am attaching the HTML with code and links to my GT3X and ActiLife raw
data output CSV files provided, which show those attempts failing.

Bests / Pozdrawiam,
Marta Karas

paulhibbing, 2019-09-19:
@martakarass I'm happy to look into this. Would you mind opening an issue on the [AGread repo](https://github.com/paulhibbing/AGread/issues) and attaching the files/links there? Unfortunately the attachments didn't come through over email (for me anyway).  We can link to the issue on this thread, for anyone who wants to follow along or join that discussion.

martakarass, 2019-09-19:
How wonderful! Thank you for your kind interest, and thanks for a
suggestion - I have just submitted an issue:
https://github.com/paulhibbing/AGread/issues/7

Bests,
Marta

closed 2021-07-15 by vincentvanhees