wadpac/oss-dev-webinar-series-pb-field #3: GitHub Tips

paulhibbing, 2019-09-19:
When I browse other repositories on GitHub, I get the sense I have very little idea of how to use GitHub to its fullest. I wondered if others would be willing to share tips here, particularly as we continue interacting through `Issues`.

My thought is, if anyone sees a way we're under-utilizing GitHub in our interactions, that might indicate that a tip could be helpful. This issue could be a good place to store those tips for easy reference.

Open to alternative suggestions.

vincentvanhees, 2019-09-19:
Great idea. Let's list this as possible webinar topic, I am going to use a specific issue label for that.

Discussion items may include:
- GitHub features to facilitate this group and features to facilitate software development projects.
- GitHub projects versus Zenhub projects
- Third party applications
- Effective ways to work with branches
- What to put in a commit message
- Working in an organisational versus personal account, pros and cons.
- What should be in the readme.md file and what should not be there.
- GitHub versus bitbucket and gitlab.

vincentvanhees, 2019-10-02:
Looks good, also:
- Point to additional resources for those interested to dive deeper into it
- Try to get some idea of how much the group already knows about git, I am guessing that experience level differ a lot.
- A live demo will be valuable.

vincentvanhees, 2019-10-07:
21:10:30	 From  John Muschelli : https://github.com/actigraph/GT3X-File-Format
21:10:44	 From  John Muschelli : https://github.com/actigraph/NHANES-GT3X-File-Format
21:11:07	 From  John Muschelli : https://github.com/paulhibbing/AGread
21:11:19	 From  John Muschelli : - that is from Paul
21:11:20	 From  Hyatt : Thanks so much
21:11:24	 From  John Muschelli : https://github.com/THLfi/read.gt3x
21:11:29	 From  John Muschelli : that’s a different one
21:24:12	 From  rubenbrondeel : I used it before, but not that much
21:29:28	 From  Tom Stewart : https://happygitwithr.com/
21:36:43	 From  John Muschelli : https://stackoverflow.com/a/2158271/2549133
21:40:48	 From  John Muschelli : https://figshare.com/
21:41:52	 From  John Muschelli : https://zenodo.org/record/1160410 - one of Vincent’s data set he released
21:44:41	 From  John Muschelli : Some of these with polysom have accelerometry: https://sleepdata.org/datasets
21:58:49	 From  mattpatterson : what is the r-package that is being discussed next meeting?
21:59:28	 From  Vincent van Hees : runstats

paulhibbing, 2019-10-11:
For internal project management, I often set up remotes on network drives. It took me awhile to figure out how to set that up, and once I found [this page](http://tony.halcyonlane.com/blog/2011/09/22/Using-git-at-work-on-a-Windows-network-drive/), I found myself coming back to it frequently until I had the workflow down.

I thought this tidbit could come in handy for others. Essentially, all you have to do is:

* Navigate to the network location and create an empty directory
    * I don't think it *has* to have the same name as your local repo (others can correct me), but that's always the name I use anyway
* Initialize an empty repo in the latter folder
    * (I usually open the folder, right click, select `Git Bash Here`, then run `git init --bare`)
* Navigate to your local repo
    * This assumes you've already set up (and even worked on) the project locally
* Define a remote that's mapped to the desired network location
    * e.g., run `git remote add origin "J:/some_path/my_repo"`
* Push to the remote, and you should be able to use it like normal from there

vincentvanhees, 2019-11-06:
I will close this issue now. The past agenda items in the readme links to this, so people will still be able to find it.

closed 2019-11-06 by vincentvanhees