wadpac/oss-dev-webinar-series-pb-field #4: Presentation R package runstats

vincentvanhees, 2019-09-19:
@martakarass I am creating this issue for your presentation of [runstats](https://cran.r-project.org/web/packages/runstats/index.html) at one of the upcoming webinar sessions.

Feel free to post relevant info, questions and other thoughts below.

martakarass, 2019-09-23:
Wonderful - thank you!
I noticed that the 2nd webinar (Oct 7) is precisely when most likely I am traveling to a conference/already at a conference. I would like to volunteer to present on the 3rd one (Nov 5) - let me know if this works!

vincentvanhees, 2019-09-24:
Sounds good to me. I will soon draft some agendas and will incorporate this.

martakarass, 2019-11-04:
I will share the slides prior to the webinar so as everybody can follow along even in case of imperfect internet connection!

martakarass, 2019-11-05:
Link to slides: 
https://docs.google.com/presentation/d/1_F9wbBBlAXMcSfcNnR4bgS5QVo5eDP3d8h7_QukknKE/edit?usp=sharing

vincentvanhees, 2019-11-06:
thanks, I see Tom added them to the [slides folder](https://github.com/wadpac/oss-dev-webinar-series-pb-field/tree/master/slides) of the repo. I will now close this issue.

closed 2019-11-06 by vincentvanhees