wadpac/oss-dev-webinar-series-pb-field #5: Data sharing

vincentvanhees, 2019-10-10:
As discussed on Monday this could possibly be a good topic for a webinar (early next year).

I imagine this could cover:
- Platforms for data sharing (Zenodo, Github, Figshare, national data services....)
- Format of data to make it findable, interoperable, accessible, re-useable (is hd5f the future or ...).
- What data to share and what not to share.
- .... ideas welcome

Anyone wants to volunteer to lead the discussion with a presentation or a demo?

muschellij2, 2019-10-10:
I will be out for this, but I would also discuss:
https://physionet.org/
and maybe https://sleepdata.org/.
I think gzipped (compression 9) CSV files are good for interoperability
(MATLAB, R, Python) but HDF5 is promising (with features such as random
access) and many do not see it going away any time soon.

Questions I always have
Does gait analysis make the data "identifiable"?

vincentvanhees, 2021-02-19:
Closing this issue now. It is cited in https://github.com/wadpac/oss-dev-webinar-series-pb-field, so people should be able to find it.

closed 2021-02-19 by vincentvanhees