wadpac/oss-dev-webinar-series-pb-field #7: Walking and/or Step detection

vincentvanhees, 2019-11-06:
At presentation of #4 yesterday it became clear that there is expertise/interest in the ground around walking/step detection from accelerometer data.

At a future webinar we may want to discuss what algorithm and software are out there for detecting walking and/or steps stratified by sensor attachment location, number and type of sensors.

informaton, 2020-02-29:
Presentation scheduled for 11 March CET 2020 by Marta and Matt.

martakarass, 2020-03-11:
Part 1/2 slides: [Precise walking strides segmentation from raw accelerometry data: ADEPT method](https://docs.google.com/presentation/d/1ThrdQnFIfRH72fkNdpaiBaxdRB2brWvRgE1cT5hlA3A/edit?usp=sharing)

informaton, 2020-03-23:
Did part 2/2 slides ever get posted?  They were good too.

vincentvanhees, 2020-03-24:
@m-patterson?

m-patterson, 2020-07-07:
Part 2/2 slides are available here - https://github.com/wadpac/oss-dev-webinar-series-pb-field/blob/master/slides/Gait%20Presentation.pptx

m-patterson, 2020-07-07:
Slide link here - https://drive.google.com/file/d/1asPFFvcTOHhTZDmeptpuc89m9futueI3/view?usp=sharing

vincentvanhees, 2021-02-19:
Closing this issue now. It is cited in https://github.com/wadpac/oss-dev-webinar-series-pb-field, so people should be able to find this issue.

closed 2021-02-19 by vincentvanhees