wadpac/oss-dev-webinar-series-pb-field #9: Combining spatial and accelerometer data

vincentvanhees, 2019-11-06:
Placeholder for Tom's talk on the 4th of December (5th for Tom).

paulhibbing, 2019-12-04:
@TheTS would you mind sharing a link or two to some studies/protocols that have used GPS for ~7d? I'm curious how others have handled issues related to short battery life.

TheTS, 2019-12-04:
@paulhibbing here are a few papers.

- A general overview of GPS in PA research [here](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC4272622/)
- One of [my papers](https://www.sciencedirect.com/science/article/pii/S1353829216304397) has a sentence in the procedures section about charging the battery

Here is a screenshot of the GPS information sheet provided to participants in one of our studies. We had to include a bit about turning off the device (for ethics), but ideally, it should be left on at all times (even overnight) to prevent people from forgetting to turn them back on!

[image]

vincentvanhees, 2021-02-19:
Closing this issue now. It is cited in https://github.com/wadpac/oss-dev-webinar-series-pb-field, so people should be able to find it.

closed 2021-02-19 by vincentvanhees