wadpac/oss-dev-webinar-series-pb-field #10: Time series segmentation

vincentvanhees, 2020-01-09:


martakarass, 2020-01-11:
Not necessarily whole time-series segmentation, but pattern segmentation from time-series we used in our recent work ([ADEPT paper](https://academic.oup.com/biostatistics/advance-article/doi/10.1093/biostatistics/kxz033/5572661?guestAccessKey=f3abdf65-a94d-4509-b7e9-cca2ff384528), [R package](https://cran.r-project.org/web/packages/adept/index.html)).

* The idea is to have a template of a pattern we want to segment from time-series. The template undergoes scaling (stretching/shrinking) and moves along the time-series signal. Pattern occurrence is identified interactively for location and scale where the template has the highest similarity with data: 

[image]

* Template may be a population-level one, or more specific, subpopulation-specific or subject-specific; multiple pattern templates may be used simultaneously. 

* We use it for individual walking strides segmentation from vector magnitude. I once used it successfully for 5x sit-to-stand identification in a side project too. 

I'd be super interested in what other people in the group have done w.r.t. time-series segmentation.

paulhibbing, 2020-01-11:
That's really cool @martakarass!

* My main experience in segmentation has been using Kate Lyden's [Sojourn method](https://www.ncbi.nlm.nih.gov/pubmed/23860415), which I've [adapted for use in youth](https://www.ncbi.nlm.nih.gov/pubmed/29135657). (It's also been [adapted for use with combined ActiGraph and activPAL data](https://www.ncbi.nlm.nih.gov/pubmed/27015380)). All are available through the [Sojourn package](https://cran.r-project.org/package=Sojourn).

* Recently I've been looking at how to analyze the performance of segmentation algorithms, since everyone seems to do it differently, often in problematic ways. I presented on this at ICAMPAM last summer and would be happy to so again.

* I've also started dabbling in change point detection. It's [been around forever](https://www.ncbi.nlm.nih.gov/pubmed/28603327), but it's just starting to [make its way into our field](https://journals.humankinetics.com/view/journals/jmpb/aop/article-10.1123-jmpb.2019-0006/article-10.1123-jmpb.2019-0006.xml) (at least as far as I know).

vincentvanhees, 2020-04-07:
My slides for later today are now in the [slides folder](https://github.com/wadpac/oss-dev-webinar-series-pb-field/blob/master/slides/vanHees_TimeSeriesSegmentation_OSSwebinarSeries.pdf).

paulhibbing, 2020-05-21:
Hi all,

For what it's worth, the transition pairing method paper was just [published online](https://doi.org/10.1123/jmpb.2019-0039).

When I presented a few weeks ago, it had been awhile since I looked the paper over, so I misspoke in saying that "ignoring" secondary transitions within the margin of error  would decrease the false positive rate. In reality, secondary transitions are labeled false positives, so the metrics are accurate. The issue is that the secondary transitions may include one that is better-timed (with respect to the criterion) than the primary transition (i.e., the true positive, which is usually defined as the first prediction to occur within the margin).

vincentvanhees, 2021-02-19:
Closing this issue now. It is cited in https://github.com/wadpac/oss-dev-webinar-series-pb-field, so people should be able to find this issue.

closed 2021-02-19 by vincentvanhees