wadpac/oss-dev-webinar-series-pb-field #11: Sensor fusion: 3 x gyroscope + 3 x accelerometer

vincentvanhees, 2020-01-09:


paulhibbing, 2020-01-11:
I do have some experience with this (see https://www.ncbi.nlm.nih.gov/pubmed/29271847), and I am building it into my dissertation. It's certainly something there needs to be broader discussion about.

vincentvanhees, 2020-01-11:
Thank's Paul, that is good to hear. From the abstract of your paper, I gather that you fused the the sensors at signal feature level. My current interest is in fusing the signals prior to feature calculation:

First deriving a fused estimate of device orientation relative to gravity. For that it needs to rely mainly on the accelerometer during static situation and more on the gyroscopes under dynamic conditions. Next step is to use the orientation to more optimally separate gravitational acceleration from movement-related acceleration. In the literature people have designed Kalman filters for this, while in the real time processing arena (drones etc) I see people favor more simplistic algorithms referred to as Complementary Filters.

What I do not find is open source  code to do those steps that is not only readable but also has a clear reference to a paper. So, I am in the process of designing and implementing my own fusion algorithm. There may be easier ways, but I quite like this route because it challenges me to understand what I do. What I miss is a good benchmark to test my algorithm, which I guess could either be in the form of a software or in the form of a data set that has reference values.

vincentvanhees, 2021-02-19:
I have now implemented some basic functionality along these lines in GGIR, and wrote a vignette about it: https://cran.r-project.org/web/packages/GGIR/vignettes/SensorFusionWithGGIR.html

The main purpose is to explore the topic. Once it matures I may want to split it off as an separate R package to ease re-use in other packages.

closed 2021-07-15 by vincentvanhees