wadpac/oss-dev-webinar-series-pb-field #12: Sharing experiences on teaching accelerometer data analysis - suggestions

R-Walmsley, 2020-07-04:
Hi everyone, 

The next webinar will be a discussion session on teaching + learning accelerometer data analysis. I'm preparing a few points to talk about, but it will be very much a discussion forum format. So I'd love to get suggestions on what the discussion could most usefully focus on. Some ideas so far: 
- Practicalities: what data to use? RStudio Cloud vs own laptops/installations etc? 
- The balance between 'hands-on' and theory
- What level of understanding of what the software is doing is required by different kinds of users? 
- What are common barriers and how do you deal with them? 

Would love to hear other ideas + also which of these you would like to focus on!!

vincentvanhees, 2020-07-08:
Great list Rosemary. Here are a few additionally suggestions:
* Are there any good open teaching materials (slides) for introduction lectures around physical behaviour assessment and accelerometry in particular? 
* What are good practical assignments. An easy but not very educational approach is to say to the students: "process this data with the software and try to understand the output". If there is more time a better approach could be to give students an example data set and ask them to process and clean the data and answer some basic research questions around data quality and interpretation. Analyse your own data could also be fun, but is less easy to control in terms of learning experience. Ideally you want them to learn about poor recordings and good recordings.

R-Walmsley, 2020-08-02:
Thank you- these are great ideas and I'll make sure to incorporate them!

R-Walmsley, 2020-08-05:
I've just uploaded the slides and notes - if you spot any errors/ places where what I typed didn't match what you said then let me know and I'll update them.

R-Walmsley, 2020-09-23:
We talked in August about freely available teaching resources, and I just came across this (enormous) [set of slides](https://playbook.dimesociety.org/), which has some good + potentially relevant content and is under a licence which means you can use and adapt it.

vincentvanhees, 2021-02-19:
Closing this issue now. It is cited in https://github.com/wadpac/oss-dev-webinar-series-pb-field, so people should be able to find this issue.

closed 2021-02-19 by vincentvanhees