wadpac/oss-dev-webinar-series-pb-field #13: High frequencies in accelerometer signals

vincentvanhees, 2020-07-20:
I posted a blog post today with my reflections about what to do with the high frequency content of an acceleration signal: https://accelting.com/updates/high-frequencies-in-an-acceleration-signal/

If you have any thoughts about this then let me know. If there are strong opposing views or interesting studies to discuss then it may be worth scheduling a webinar about it.

paulhibbing, 2020-07-24:
This seems important to revisit as the field transitions away from activity counts, and on towards the use of raw acceleration signals. In my mind, filtering had a different significance for calculating counts than it does for processing raw acceleration data, and in light of that transition it would be a good idea to discuss this as a field directly, rather than the indirect comments it's received in the past.

Some scattered thoughts:

* I found three reviews that gave some references for previous research on the range of frequencies generated by human motion. There seem to be places where the evidence overlaps, and places where it doesn't.
  1. [Mathie et al., 2004](https://iopscience.iop.org/article/10.1088/0967-3334/25/2/R01/meta?casa_token=uRj3totAaKoAAAAA:zU6Tj5ymBfvJ8fwfEGKi1-2aNrnqQrtZ1ht5aEHNCzR29_No63CzWmuA92vYhkzVBNgDTYM-)
  2. [Esliger et al., 2005](https://journals.humankinetics.com/view/journals/jpah/2/3/article-p366.xml)
  3. [Migueles et al., 2017](https://link.springer.com/article/10.1007/s40279-017-0716-0)

* It's interesting to note that previous research in this vein was necessarily conducted using raw data, even when counts were the predominant unit of application. That just exacerbates the disconnect between raw/counts.

* We should also discuss how filtering interacts with segmentation and downsampling. It's one thing to filter an entire waveform and then downsample it (say, by averaging each second, which is common). It would be another to downsample it and then filter it (which I don't think would be appropriate, but may not be much different from how frequency domain features are sometimes extracted for various models). And it would be still another to perform some initial pre-processing (e.g. segmentation and pre-classification) of the unfiltered data, followed by segment-specific filtering to gain more detail and make a final prediction.

* I think sometimes a utilitarian approach can be justified when it comes to filter cutoffs (or lack thereof). For example, some of our work with gyroscopes has used different low-pass cutoffs. For [turn detection](https://iopscience.iop.org/article/10.1088/2057-1976/aad0d0/meta?casa_token=xx2rgXfuTngAAAAA:XgyVGHpLeuVaRKG5p_p6jscJVgNVniJj_CLw0W6sXGU7lxHjqBd6n1Fvwh2y9QhsR8zX9-Q7), Rob Marcotte found he got the best results using a very low cutoff, whereas for [energy expenditure estimation](https://europepmc.org/article/med/29271847) we used a 35-Hz cutoff (although that was based largely on what conventions we could already find in the literature, of which there were a handful).

vincentvanhees, 2020-07-27:
The oldest statement about this topic is probably from Carlijn Bouten in the 1990s, one of the accelerometer pioneers. Her thesis is available [here](https://research.tue.nl/en/publications/assessment-of-daily-physical-activity-by-registration-of-body-mov),  [link to pdf](https://research.tue.nl/files/2262547/443690.pdf), and on page 34 she cites ([Antonsson & Mann, 1985](https://pubmed.ncbi.nlm.nih.gov/3980487/); [Sun & Hili, 1993](https://www.sciencedirect.com/science/article/abs/pii/002192909390361H) to support her statement that human movement cannot produce frequencies beyond 20 Hertz. This statement is repeated many times throughout her thesis. So, clearly she was very certain about it. I do not have access to those papers from where I am, but it looks like these were experimental studies.

I wonder what kind of experiment would allow us to demonstrate the relative contribution of harmonics versus noise in an acceleration signal. My gut feeling is that the only way to separate noise from movement is to know the true acceleration. Like the artificial pendulum motion I use in my blog post. As soon as we start looking at real world data there is immediately the uncertainty about whether high frequencies may also reflect noise of some kind. A robot experiment may be able useful but then we would have to be 100% certain that the robot itself does not produce noise.

Interestingly many of the researchers who have made statements about the need for filtering were qualified engineers or computer scientists by background. So, it is not like these people lacked the background to face this challenge .... or maybe they all lacked the same tiny area of expertise around time series harmonics. Another theory could be that they all worked in an inter-disciplinary research environment where it was tempting to accept the beliefs of some human movement researcher that human movement only occurs at frequencies below 20 Hertz.

paulhibbing, 2020-07-29:
Pragmatism may have come into play as well. There are many sources of error in activity measurements, and filtering may not have appeared worth fine tuning. Obviously there was some need for it (and ActiGraph's low frequency extension implies that the gross settings matter, at least for the low end of a bandpass filter), but perhaps minor manipulations didn't seem to make large differences. I can't necessarily speak to that, but I will say again that I think most perceptions in this vein are based on how filtering affects activity counts, subsequent cut-point-based predictions, and step counts. It does seem worth reopening the discussion with an emphasis on raw acceleration and subsequent predictions, perhaps from machine learning models rather than cut-points.

This seems like something Daniel Arvidsson and Jan Brønd might have good insights on.

vincentvanhees, 2020-07-29:
Last week I reached out to Jan about this topic to get his opinion. I have not received a reply (yet). 

I cannot remember that Jan studied acceleration signal noise specifically in any of his publications, but I do remember that he is generally keen to discuss acceleration signal filter related issues. I decided that a blog post with my own perspective and some attempts to start informal discussions would be a good way to push for more insight.

The lower end of the band-pass filter should I think be treated as a separate discussion. There it the issue is less about noise or harmonics and more about how to separate the gravitational component from movement.

m-patterson, 2020-07-29:
Very interesting discussion. Two important considerations are the mounting location of the sensor as well as the desired end-point. I would expect that a sensor on the wrist will be exposed to higher movement frequencies than a sensor mounted on the waist or trunk. I have not come across any work that attempted to quantify these differences. The desired end-point is also relevant as detecting physical activity in an elderly population will likely require a lower low pass filter compared to trying to measure jump height (or peak velocity during jumping) from an elite athlete.

The god-father of biomechanics, David Winter used the residual analysis technique to find the optimal cut-off frequency for marker data from walking. The research found that the toe and heel trajectories had the highest harmonics of all markers placed on the lower limbs during walking. The research also found that 99.7% of the signal power was contained in the lower seven harmonics, which were below 6-hz. Above the seventh harmonic, there was till some signal power, but it had the characteristics of noise. Noise in this case was determined by differentiating the data twice to obtain acceleration [[Winter, 2009](https://edisciplinas.usp.br/pluginfile.php/4174628/mod_resource/content/2/David%20A.%20Winter-Biomechanics%20and%20Motor%20Control%20of%20Human%20Movement-Wiley%20%282009%29.pdf) – Section 3.4.2 to 3.4.4]. 6-hz became the industry standard for filtering marker data for walking applications.

Perhaps, a similar exercise could be used to identify noise in an acceleration signal. Perform double integration to obtain displacement (over a short period of time) and compare to a known displacement (ie, from a motion capture marker in the experiment). What cut-off frequencies are resulting in ideal displacement data? The issue I do see with this, is that displacement is not the goal of activity monitoring algorithms, the goal is to quantity physical activity levels.

Soft tissue artefact and sensor mounting artefact are definitely problems with accelerometer data. Probably more of an issue in remote monitoring scenarios, where patients are required to put on the sensor by themselves, thus more risk for a loose attachment than if a researcher put the sensor on for them. Even in motion capture, soft tissue artefacts are a concern. Researchers have gone as far as to surgically implant bone screws into subjects lower legs to investigate this.

Soft tissue artefact and marker movement are big issues with optical motion capture systems. The goal with optical motion capture is to measure the skeletal movement of a person. Sensor shaking due to soft tissue artefact affects this, as well as sensor shaking due to poor mounting techniques. Researchers have gone as far as to surgically insert screws into the bones of the lower legs, mount markers on the screws and compare them to markers mounted on the skin next to the screws to quantify the effect that soft tissue artefact has on calculating skeletal movement [[Benoit et al, 2006](https://www.sciencedirect.com/science/article/abs/pii/S0966636205001700)].

paulhibbing, 2020-07-29:
I'd forgotten about residual analysis. Dunno if it's helpful, but a few years ago I looked into that for finding a good low-pass cutoff for gyroscope data being used to predict energy expenditure. At the same time Rob Marcotte was looking into gyroscope filtering for turn detection (referenced in a previous comment).

The residual analysis came out with something low, like 8 Hz. But for energy expenditure we ended up going with a much higher value (35 Hz) from the literature. And for turn detection Rob had the greatest success quite a bit below 8 Hz.

None of that means we did things exactly right, but at a minimum it seems to me there are a range of goals that might influence how filters are approached field-wide, including:

* Event detection
* Physical activity assessment
* Instrument fidelity for its own sake

On the last note, greater fidelity would theoretically translate to improved downstream measurements. But there may be some debate around how great the improvements would be, and that may (again) differ depending on what end users are hoping to achieve.

I also wonder if there is a perception that an optimal cutoff is a hard line between signal and noise, rather than a tradeoff between preserving as much signal as possible and eliminating as much noise as possible. (Or maybe I'm the one with misconceptions...)

vincentvanhees, 2020-07-29:
Great suggestions, thanks both. All of this actually connects with our earlier conversations about selecting the sample rate. Lowering the sample rate is a very drastic form of removing higher frequencies. So, maybe we can find additional answers by checking out studies that have tried to provide guidance on sample rate selection.

closed 2021-07-15 by vincentvanhees