wadpac/oss-dev-webinar-series-pb-field #15: NHANES accelerometer data analysis

informaton, 2020-09-01:
This is for the webinar on 9/2/2020.    

Here is the working draft of my [google slides](https://docs.google.com/presentation/d/e/2PACX-1vRUZdwJEA3K0XhdlJ5_C2BpoI2xaqYx8bE9n3yQAT1dHcvYlwhkTE_AcByCe8n33OJiithgPUmtd4iM/pub?start=false&loop=false&delayms=3000) for this presentation.

informaton, 2020-09-01:
### Manuscripts

* Troiano, Richard P., et al. "[Physical activity in the United States measured by accelerometer.](https://journals.lww.com/acsm-msse/Fulltext/2008/01000/Physical_Activity_in_the_United_States_Measured_by.25.aspx)" Medicine and science in sports and exercise 40.1 (2008): 181.
* Mooer, Hyatt E., et al. "[Potential corner case cautions regarding publicly available implementations of the National Cancer Institute’s nonwear/wear classification algorithm for accelerometer data.](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC6312247/)" PLOS ONE 13.12 (2018): 1-11.

### NHANES Datasets
* NHANES Questionnaires, Datasets, and Related Documentation ([CDC website](https://wwwn.cdc.gov/nchs/nhanes/Default.aspx))
* NHANES 2003-2004 links ([CDC link](https://wwwn.cdc.gov/nchs/nhanes/continuousnhanes/default.aspx?BeginYear=2003))
* NHANES 2003-2004 physical activity monitor data dictionary (([PAWRAW_C.htm](https://wwwn.cdc.gov/Nchs/Nhanes/2003-2004/PAXRAW_C.htm)))

### Implementations of NCI's nonwear algorithm

#### National Cancer Institute's SAS code
* Website: 	https://epi.grants.cancer.gov/nhanes_pam/create.html
* Macro (/day):	https://epi.grants.cancer.gov/nhanes_pam/create.pam_perday.sas 

#### R Accelerometry project 
* Website: 	https://cran.r-project.org/web/packages/accelerometry/index.html
* Doc (.pdf): 	https://cran.r-project.org/web/packages/accelerometry/accelerometry.pdf

#### ActiLife
* White paper:	[What is the difference between the wear time validation algorithms](https://actigraph.desk.com/customer/en/portal/articles/2515570-what-is-the-difference-between-the-wear-time-validation-algorithms- )

dijunrui, 2020-09-01:
FYI, there's a processed data package for NHANES 0304/0506 PA data, developed by Andrew Leroux. 
https://oslerinhealth.org/package/rnhanesdata

vincentvanhees, 2021-02-19:
Closing this issue now. It is cited in https://github.com/wadpac/oss-dev-webinar-series-pb-field, so people should be able to find this issue.

closed 2021-02-19 by vincentvanhees